AutonomyLab/ardrone_autonomy #130: Problem with the GPS data

shehroze37, 2015-01-18:
Hi I am trying to get the GPS data from the parrot.Previously I was using ardrone on Ubuntu 14.04 with ROS indigo and because GPS was only supported in the hydro branch, I installed hydro on UBUNtu 12.04 but it still says that topic ardrone/navdata_gps has not been published yet ??

In the GPS branch I could not get ardrone autonomy using the following command apt-get install ros-hydro-ardrone_autonomy so instead I usedapt-get install ros-hydro-ardrone-autonomy. The error generated was that package ros-hydro-ardrone_autonomy cant be found 

Whats going wrong here?
Is there any way I could get GPS on Ubuntu 14.04 with ROS indigo ??

I have spent so much time on this and would really appreciate any help 

Thanks

KekkeG, 2015-02-09:
I'm also interested in GPS support when running under Indigo. I would really appriciate if GPS support became available. Even with bugs, I can fiddle with the bugs. 

Anyway, thanks for this package :)

JCUAV, 2015-10-09:
I answered myself here. http://answers.ros.org/question/218846/receiving-no-data-from-ardronenavdata_gps-topic/?answer=218903#post-id-218903

mani-monaj, 2015-10-09:
For GPS support you need to compile the `gps-waypoint` branch of the driver from source. More infromation:

- http://ardrone-autonomy.readthedocs.org/en/latest/FAQ.html#is-there-any-support-for-gps-parrot-flight-recorder
- http://ardrone-autonomy.readthedocs.org/en/gps-waypoint/

closed 2015-10-09 by mani-monaj