AutonomyLab/ardrone_autonomy #136: Dont know why cannot read navdata

erivera1802, 2015-03-30:
Hi,
I really hope you can help me, recently I installed the ardrone_autonomy package, but I havent make it work. 
Mi first problem is that I cannot read the navdata topic:
edrone@edrone-Lenovo-IdeaPad-Y410P:~/catkin_ws/devel$ rostopic echo /ardrone/navdata 
ERROR: Cannot load message class for [ardrone_autonomy/Navdata]. Are your messages built?

Also, I try to paint some of the data from navdata with rq_plot, but it also doesnt work.
I apreciate your help,
Esteban Rivera

erivera1802, 2015-04-03:
Hi,
I could read the data, but the linear z velocity is steady at 0. Only this value is wrong, could you tell me what can i do to solve this issue?
Thanks

feixia586, 2015-07-22:
Hi, @erivera1802  I have the same problem:

$ rostopic echo /ardrone/navdata 
ERROR: Cannot load message class for [ardrone_autonomy/Navdata]. Are your messages built?

How did you manage to solve this? Thank you!

mani-monaj, 2015-08-07:
> ERROR: Cannot load message class for [ardrone_autonomy/Navdata]. Are your messages built?

When compiling the driver from the source, you need to source the `devel/setup.[bash|zsh]` file of the workspace in any new terminal window that interacts with the driver. This is not required when you install the binary version of the driver.

>  could read the data, but the linear z velocity is steady at 0

Where do you get this value from? `Navdata` or `Odometery`?

feixia586, 2015-08-07:
Problem solved! Thank you!

closed 2015-08-08 by mani-monaj