AutonomyLab/ardrone_autonomy #139: Publish TF at the same rate as odometry

v01d, 2015-05-14:
Currently TF transforms are sent at a fixed (and low) rate, which introduces big (and unnecessary) delays on the receiving when waitForTransform is used. Moreover, since the tf_odom transform is tied to the data that is read on publishOdometry, it makes more sense to publish this directly from this function. 

The fixed transforms (for both cameras) could still be published in the current function since these are not expected to change. However, publishing tf_odom from PublishOdometry makes more sense.

I can provide a pull request but the change is trivial (move the publish call to the end of PublishOdometry).

mani-monaj, 2015-05-15:
That's a valid point. A PR is certainly welcome and appreciated. I set the milestone to 1.4.1.

mani-monaj, 2015-08-11:
Fixed on #144

closed 2015-08-11 by mani-monaj