AutonomyLab/ardrone_autonomy #140: Change camera frequency

baryn, 2015-05-22:
Hi, 
I am working with the bottom camera. I control the AR Drone 2 by sending command velocities which are computed from information of the video stream (basically features coordinates). Meaning I send the commands at the camera frequency because the velocities need to be updated at each new frame, which is about 10Hz (http://ardrone-autonomy.readthedocs.org/en/latest/frames.html) and it is not fast enough actually. So, I wondered if it was possible to higher this rate, indeed it is a 60fps camera.
I use a code similar to the following one: http://wiki.ros.org/cv_bridge/Tutorials/UsingCvBridgeToConvertBetweenROSImagesAndOpenCVImages

Thanks a lot
baryn

baryn, 2015-06-02:
Hi,
I found my problem, but yet, not the solution. I have a delay of more than 1sec betweem the velocity I send and the drone's response. I thought it was due to a low control rate depending on the rate of the camera. It is not due to a delay induced by my code or the publishing because I plotted the IMU pitch and with a sinusoidal velocity input it follows the input only 1 or 2 iterations later (50ms to 100ms at a rate of 20Hz). So, I believe it is due to the drone or more specifically to its PID. Now, is it possible to have access to this PID ?

Thanks,
baryn

kbogert, 2015-06-02:
The velocity command's linear.x and y are actually pitch or roll commands, you will need to provide your own PID if you are trying to reach a target velocity. See: #116

baryn, 2015-06-02:
Wow, really ?! That explains my problem. It wasn't obvious I guess, I must have missed something. Anyway, thank you very much. Issue solved !

closed 2015-06-03 by baryn