AutonomyLab/ardrone_autonomy #141: How to make ardrone move continuously for 2 seconds

saikishor, 2015-05-28:
My problem is:
i want to move ardrone forward for 2 seconds and then stop. So, i can make it move forward by making linear.x +=1 and then so but how to make it for 2 seconds and then stop.

I tried using time.sleep() for 2 seconds. it works only for 4 times out of 10 trails please help me
problem is like its not stopping after 2 sec. I tried making that linear.x value as 0

kbogert, 2015-05-28:
The linear.x and y are not velocity commands, they control the rotation angles of the drone.  (see #116) Your options are:

1.  Send a smaller value for linear.x, use ROS's sleep [http://wiki.ros.org/roscpp/Overview/Time] for 2 seconds, and then send a full set of zeros to cmd_vel to make the drone hover.  This will sort of work.
2.  Integrate the linear.x command to get to the velocity you want, then send negative values to slow the drone down, finally giving it full zeros to hover.
3.  Use the tum_ardrone or similar that does state estimation and a PID controller to control the drone

saikishor, 2015-05-28:
Ohh great thanks actually i found where the problem is?
there is a problem in publishing commands to drone. Now it is cleared... Thanks a lot @kbogert

closed 2015-05-28 by saikishor