AutonomyLab/ardrone_autonomy #142: Problem in AR. Drone Movement (angular.z)

saikishor, 2015-05-28:
The problem with AR.Drone is
when i give a command of linear.x = +1 for 2 seconds and then making it zero. When, i repeat this as a sequence of 10, AR.Drone is getting out of track i.e., its angular.z is changing as it moves and it is getting deviated out of the path. Is there any solution for this problem

mani-monaj, 2015-05-30:
Without proper feedback (via state estimation) it's impossible to bound the drone's movement error. You can calibrate the magnetometer using _FreeFlight App_ to reduce the yaw error for a better open loop performance.

Take a look at [tum_ardrone](http://wiki.ros.org/tum_ardrone) or [ar_track_alvar](http://wiki.ros.org/ar_track_alvar) for vision based state estimation solutions. The former comes with a PID controller for waypoint navigation. You need to implement your own if you use the latter.

Since v1.4, the driver publishes _Odometery_ data by integrating on-board velocity estimates. You can use that as a crude feedback. However, due to the nature of this feedback signal, it is imprecise and subject to drift.

saikishor, 2015-06-03:
Wow!!! thanks it helped me @mani-monaj

closed 2015-06-03 by saikishor