AutonomyLab/ardrone_autonomy #145: AR.Drone 2.0 altitude from navdata is showing wrong value

saikishor, 2015-06-03:
I am having a problem. This issue is very important to me. If anyone helps me in this, i can complete my project.
 
The problem is:
When my ardrone is flying, I will be able to get altitude value from /ardrone/navdata/altd and then when the ardrone moves and when we find it above a table of height x and then normally the altitude of the drone should be previous - x, but this is not happening. Why??

Example :  when its flying, let's say the altitude be 800 mm
and then, table height be 300 mm 
so when the ardrone is above the table, then altitude should be 800-300 = 500mm but it is not working such a way.

alexjung, 2015-08-06:
This is actually not an issue. The altitude from /ardrone/navdata/altd is a measurement that combines multiple sensor inputs and not only the sonar height above ground. You need this altitude measurement so your drone doesn't jump up if it hovers over a table like you described it.

But you can get the height measured from the ultrasonic sensors by enabling it in the advanced navdata. I have done so by setting the bool in NavdataMessageDefinitions.h in line 259 to true. Line looks now like this: ros::param::param("~enable_navdata_altitude", enabled_navdata_altitude, true);

Hope I could help!

mani-monaj, 2015-08-07:
@alexjung Thank you for your answer.

@saikishor Please re-open the issue if you still have the problem.

closed 2015-08-07 by mani-monaj