AutonomyLab/ardrone_autonomy #149: when I run ardrone_driver,I get Segmentation fault.how can i fix it?

csc301, 2015-06-12:
I use ubuntu14.04.02 and ros-indigo,I download the ardrone_autonomy and complie it ,it is ok .but when I run the ardrone_driver.I get Segmentation fault?

Napoleon-BlownApart, 2015-06-15:
I have the same setup, and mine works ok.  Can you supply more info?
How you are launching it, do you have any ROS::Params set? etc.

csc301, 2015-06-18:
I connect my ardrone2.0 to my laptop and run the ardrone.launch in the ardrone_autonomy/launch directory with the default parameters,then I get segmentation fault.But today I delete the workspace and make a new one,then git clone your ardrone_autonomy package and catkin_make it.when I run the driver launch and it is OK now.   Sorry for the trouble and think you very much.

Phelipea, 2015-07-16:
Didn't work for me, i'm using ubuntu 12.04 and the hydro version of ROS. With the package install in my workspace it's now possible to run the ardrone_driver node, neither through the launch file of course. Sadly it only works when i make the binary install, the problem is that i need to modify the video.h file and it doesn't exists in my /opt directory. Any thoughts?

mani-monaj, 2015-08-07:
@Phelipea 

>  With the package installed in my workspace it's not possible to run the ardrone_driver node

Could you please paste the error you get here?

Flaret, 2015-10-23:
Dear all,

I have the same issue.
_Test system:
ARDrone 1.0
Ubuntu 14.04 LTS with latest updates. (On the PC an in the VirtualBox)
Latest stable ardrone_autonomy package (1.4.1 (2015-10-18))

//=================================================================
:~$ rosrun ardrone_autonomy ardrone_driver 
Wait authentification
Wait authentification
Wait authentification
===================+> 192.168.1.1
[ INFO] [1445595423.678223315]: No values found for `~cov/imu_la` in parameters, set all to zero.
[ INFO] [1445595423.680088187]: No values found for `~cov/imu_av` in parameters, set all to zero.
[ INFO] [1445595423.681553443]: No values found for `~cov/imu_or` in parameters, set all to zero.
[ WARN] [1445595423.689466530]: realtime navdata is off, odometry may be imprecise
Deleting Profile -cea48d22
Deleting Application -e182b69b
Input device Teleop added
Starting thread video_stage
Starting thread update_ros
Set IP_TOS ok
Starting thread navdata_update
Starting thread ardrone_control
PA : MEMORY SPACE ALLOWED : 40 MB
Start thread thread_academy_upload
Start thread thread_academy_download 
Academy download stage resumed
Start thread thread_academy
Thread navdata_update in progress...
Thread `update_ros` started 
 
video stage thread initialisation

Video multisocket : init 2 sockets
Video multisocket : connecting socket 0 on port 0 UDP
Video multisocket : connecting socket 1 on port 5555 TCP
 - Connection failed
Academy download stage paused
Timeout when reading navdatas - resending a navdata request on port 5554
Timeout when reading navdatas - resending a navdata request on port 5554
Timeout when reading navdatas - resending a navdata request on port 5554
Timeout when reading navdatas - resending a navdata request on port 5554
Timeout when reading navdatas - resending a navdata request on port 5554
[ INFO] [1445595428.819750540]:   SEND: CAT_APPLI/navdata_options = 268435455.000000 (DEFAULT = 65537.000000)
Segmentation fault (core dumped)

//=================================================================

mani-monaj, 2015-10-23:
@Flaret It might be an issue with networking inside your virtual machine. Have you tried running the driver on a non-virtualized Ubuntu?

Flaret, 2015-10-23:
@mani-monaj I have both identical Ubuntu configurations: real and virtual. And both give me the same error.

mani-monaj, 2015-10-23:
@Flaret Ok. It's been a while since last time we tested the driver with AR-Drone 1.0. Could you please also provide the firmware version of your AR-Drone?

Flaret, 2015-10-23:
@mani-monaj It runs BusyBox v1.14.0. Linux version 2.6.27.47-parrot-01227-g93dde09 (fdhaeyer@busard) (gcc version 4.2.2). Is there any way to get drone firmware version without mobile?

mani-monaj, 2015-10-25:
@Flaret You may be able to `telnet` into the drone and look at the files in `/data/` folder. I am not sure about it.

Flaret, 2015-10-28:
@mani-monaj Mobile gives version of ARDrone frw.: 1.1.1
But reading /data/config.ini gives: 
num_version_soft = 1.1.3
soft_build_date = 2010-08-05 12:26

mani-monaj, 2015-10-28:
@Flaret I think the firmware is way too old. I did a quick search and it seems like the latest AR-Drone 1.0 firmware version is `1.7.x`. Please update the firmware and see if you can get the driver working with your AR-Drone.

http://ardrone2.parrot.com/update-ardrone-1/
http://drone-apps.com/support/ar-drone-firmware/

Flaret, 2015-10-28:
@mani-monaj Maybe firmware is old but on the ardrone_autonomy from October 22 2013 it works fine.

Flaret, 2016-01-19:
@mani-monaj It was a long time since I was absent.
Before doing firmware update I would like to debug errors.

I updated **ardrone_autonomy** package with *git pull*.

/**log before the error**=====================================================/
roslaunch ardrone_autonomy ardrone.launch 
... logging to /home/gici/.ros/log/c952c7f8-be00-11e5-a401-00224d561227/roslaunch-PC-GICI-2-634.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://PC-GICI-2:49002/

SUMMARY
========

CLEAR PARAMETERS
 * /ardrone_driver/

PARAMETERS
 * /ardrone_driver/altitude_max: 4000
 * /ardrone_driver/altitude_min: 50
 * /ardrone_driver/bitrate: 4000
 * /ardrone_driver/control_vz_max: 700
 * /ardrone_driver/control_yaw: 1.75
 * /ardrone_driver/cov/imu_av: [1.0, 0.0, 0.0, 0...
 * /ardrone_driver/cov/imu_la: [0.1, 0.0, 0.0, 0...
 * /ardrone_driver/cov/imu_or: [1.0, 0.0, 0.0, 0...
 * /ardrone_driver/detect_type: 10
 * /ardrone_driver/detections_select_h: 32
 * /ardrone_driver/detections_select_v_hsync: 128
 * /ardrone_driver/enemy_colors: 3
 * /ardrone_driver/enemy_without_shell: 0
 * /ardrone_driver/euler_angle_max: 0.21
 * /ardrone_driver/flight_without_shell: 0
 * /ardrone_driver/max_bitrate: 4000
 * /ardrone_driver/navdata_demo: 0
 * /ardrone_driver/outdoor: 0
 * /ardrone_driver/realtime_navdata: True
 * /ardrone_driver/realtime_video: True
 * /ardrone_driver/ultrasound_freq: 8
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    ardrone_driver (ardrone_autonomy/ardrone_driver)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[ardrone_driver-1]: started with pid [653]
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "i686-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/gici/catkin_ws/devel/lib/ardrone_autonomy/ardrone_driver...done.
Starting program: /home/gici/catkin_ws/devel/lib/ardrone_autonomy/ardrone_driver __name:=ardrone_driver __log:=/home/gici/.ros/log/c952c7f8-be00-11e5-a401-00224d561227/ardrone_driver-1.log
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[... 125 lines cut ...]
        __res = <optimized out>
        pd = 0xae9feb40
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {-1210851328, -1365251264, 4001536, 
---Type <return> to continue, or q <return> to quit---return
                -1365253144, -1888545718, 366414456}, mask_was_saved = 0}}, priv = {pad = {
              0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#3  0xb7b16bee in clone () at ../sysdeps/unix/sysv/linux/i386/clone.S:129
No locals.
/=======================================================================/

/**bt no-filters**============================================================/
(gdb) bt no-filters
#0  0x0812831b in ARDroneDriver::run (this=0x85d9e78)
    at /home/gici/catkin_ws/src/ardrone_autonomy/src/ardrone_driver.cpp:146
#1  0x08192068 in thread_update_ros (data=0x85d9e78)
    at /home/gici/catkin_ws/src/ardrone_autonomy/src/ardrone_sdk.cpp:55
#2  0xb7d2bf70 in start_thread (arg=0xae9feb40) at pthread_create.c:312
#3  0xb7b16bee in clone () at ../sysdeps/unix/sysv/linux/i386/clone.S:129
/========================================================================/

Flaret, 2016-01-27:
@mani-monaj 
Do you know if the latest ARDrone 1.0 firmware (1.11.5) supports SDK 2.0.1?
Maybe you can suggest which version better to use?

I've managed somehow to run ardrone_autonomy on ARDrone 1.0 with firmware 1.1.3 and SDK 2.0.

Flaret, 2016-01-31:
@mani-monaj 
Update to the latest firmware fixed all issues.
I think that this topic could be closed.

mani-monaj, 2016-02-02:
@Flaret great.

closed 2016-02-02 by mani-monaj