AutonomyLab/ardrone_autonomy #152: Using tag_type

Himanshirocks, 2015-06-28:
I would like to track an object using the ardrone but first I wanted to try tracking the oriented roundel or a colored marker with the help of tag_type.

But I don't know exactly how to proceed. Can somebody give any pointers or related links? I couldn't find anything on other platforms.

raultron, 2015-07-22:
Hi, I have the same problem. Did you find a solution?.

raultron, 2015-07-23:
I already found how to do it. The answer is in the Ar.Drone Developer Guide, at the end it describes all the parameters, each of those parameters correspond to the parameters in the ROS ardrone_autonomy package. Maybe the documentation of the ardrone_autonomy should remark that the complete description of the parameters can be found in the Ar.Drone Developer Guide.

If somebody else wants to detect the oriented roundel with the bottom camera and detect the color stripes with the front camera make a launch file with this parameters:

<param name="detect_type" value="10" />
<param name="enemy_colors" value="3" />
<param name="detections_select_h" value="32" />
<param name="detections_select_v_hsync" value="128" />
<param name="enemy_without_shell" value="0" />
 
<param name="enable_legacy_navdata" value="true" />
 
<param name="enable_navdata_demo" value="true" />
 <param name="enable_navdata_time" value="true" />
<param name="enable_navdata_altitude" value="true" />
<param name="enable_navdata_vision_raw" value="true" />
<param name="enable_navdata_vision_of" value="true" />
<param name="enable_navdata_vision" value="true" />
<param name="enable_navdata_vision_perf" value="true" />
<param name="enable_navdata_trackers_send" value="true" />
<param name="enable_navdata_vision_detect" value="true" />

Then in the topic navdata_vision_detect you will have the detection information.

Bests,

mani-monaj, 2015-08-07:
Would you mind creating a self-contained launch file to show case your solution and send a pull request?  I will be more than happy to merge that into the next release.

closed 2015-08-07 by mani-monaj