AutonomyLab/ardrone_autonomy #153: Periodically reduced sampling time

Alemouw, 2015-07-14:
Hi! I'm trying to use ardrone_autonomy on ROS (indigo, running on Ubuntu) to get measurements from a Parrot AR Drone 2.0 via Wi-Fi. However, I've noticed an interesting (and annoying) phenomenon.

The samples are supposed to arrive at a rate of about 200 Hz, and most of the time this is the case. However, once every 1-2 minutes, the sampling rate is reduced to only about 8-20 Hz. This condition lasts for around 6 seconds before returning to normal.

I've tried with two different quadcopters, but the problem persists. I've also changed the desired sampling rate, but the problem remains. It only "disappears" when you choose a sampling frequency lower than about 5 Hz, meaning the problem is probably still there, just not detectable.

Attached is a picture of the raw data of a recorded .bag file visualized in rqt, where blue means a sample was received and white means no sample was received. The data from the ardrone/navdata topic looks identical.

[image]

Any help in troubleshooting would be appreciated!
Thanks in advance!

mani-monaj, 2015-08-07:
Have you tried to run the driver on another machine? My guess is this issue is due to some network problems. Are you running the driver on a laptop? If yes, are you using the internal wireless card to connect to the drone?

Alemouw, 2015-08-07:
I have not tried the driver on another machine, as I currently only have access to one. It's a laptop and yes, I'm using its internal wireless card for the connection.

Thanks for taking the time to comment! :)

Alemouw, 2015-08-07:
It might be worth noting that as far as I've experienced, this does not seem to occur when connected to other networks, such as internet connections etc. Though this may simply be due to me failing to notice.