AutonomyLab/ardrone_autonomy #163: Timeout when reading navdatas

SergioGarG, 2015-10-03:
Hello,

I got a problem running the ardrone_autonomy package. I’m running it with Ubuntu 14.04 and indigo. After writing the following on a terminal:

roscore
rosrun ardrone_autonomy ardrone_driver (in a new terminal)

The system texts me back this:

Timeout when reading navdatas - resending a navdata request on port 5554
Reconnecting ... OK
Reconnecting ... OK
Reconnecting ... OK
Timeout when reading navdatas - resending a navdata request on port 5554
Reconnecting ... OK
Reconnecting ... OK
Reconnecting ... OK
Timeout when reading navdatas - resending a navdata request on port 5554
Timeout when reading navdatas - resending a navdata request on port 5554
Timeout when reading navdatas - resending a navdata request on port 5554

I’ve tried following the advices given in:

https://github.com/AutonomyLab/ardrone_autonomy/issues/70

And here:

https://github.com/AutonomyLab/ardrone_autonomy/issues/118

My laptop is connected only to the ardrone's wifi network and I can telnet (192.168.1.1) to it. I’ve also tried connecting it with wpa_supplicant, but with the same result.

I’m thinking about buying a wifi USB adapter because at this point I don’t know if it could be a problem related to my wifi card (could it be?). I’ve never have issues connecting my laptop to wifi networks, but other computers of the laboratory can connect to the ardrone’s wifi network without problems. I have the same problem connecting the laptop to the bebop’s wifi network (I mention it because I don’t know if it could help).

Please, can anyone help me?

Thanks in advance,
Sergio

raultron, 2015-10-06:
I had similar problems and it was related to the driver of my Wifi dongle. I had to find a custom driver in Github and compile the module for my kernel. After doing this I have no more problems.

I also remember that I changed the Channel of my wifi card to one less congested.

Cheers,

SergioGarG, 2015-10-06:
Yes, it was the problem. I also tried with an external wifi adapter an it is working.

Thank you.

closed 2015-10-09 by mani-monaj