AutonomyLab/ardrone_autonomy #165: Set covariance matrices in ardrone/odometry

steinio, 2015-11-02:
Right now the covariance matrices of this topic is zero by default, while the covariance of the other topic, ardrone/imu, is set as a parameter in the launch file at startup. So instead of having to republish the data continuously with given covariance values (only solution i could think of), is it possible to mae it so that we can set the ardrone/odometry covariance matrices in a similar fasion as with ardrone/imu? 

My first issue ever, so i apologise if this is the wrong way of going about this.
I asked about this at answers.ros aswell, and someone told me to make an issue here.

mani-monaj, 2015-11-03:
These are the minimum changes that needs to be made to  [ARDroneDriver::PublishOdometry](https://github.com/AutonomyLab/ardrone_autonomy/blob/2e3b75a911e922dbaecbd35609cb1f46f6f65618/src/ardrone_driver.cpp#L684) member function and its parent class.

1. `odo_msg` should become a private member of ARDroneDriver.
2. In the constructor of `ARDroneDriver`, constant covariance values of `odo_msg` should be fetched from params using the helper function [ReadCovParams](https://github.com/AutonomyLab/ardrone_autonomy/blob/2e3b75a911e922dbaecbd35609cb1f46f6f65618/src/ardrone_driver.cpp#L232) and set accordingly. 

Please feel free to implement this and submit a pull request. If no PR is submitted, we will implement this feature as part of `1.5` release.

mani-monaj, 2015-12-22:
@steinio Any luck with implementing these changes?

jbler, 2016-04-07:
I implemented these small changes to allow to set the covariance matrices of odometry pose and twist in the launch file. I followed your guideline implement the modifications.

The covariance matrices of odometry are separated in two matrices of size 6x6 = 36 entries. However, the imu covariance matrix is build up of sub-matrices with 9 entries. Therefore I modified the ReadCovParams helperfunction.

1. I set odo_msg to a private member of ARDroneDriver
2. I modified [ReadCovParams](https://github.com/jbler/ardrone_autonomy/blob/master/src/ardrone_driver.cpp#L242) to handle the fixed size of boost::array

It is worth notable, that values which are referenced in 1eX representation will lead to an error in the string stream function. 

It is my first contribution, so I am not sure if I am doing something wrong. I will send you an pull request.

mani-monaj, 2016-04-07:
Thank you for taking the time to implement this. Let's continue the discussion on #182

closed 2016-04-07 by mani-monaj