AutonomyLab/ardrone_autonomy #166: Loading parameters in adrone_driver

joseparnau, 2015-11-06:
Hi everyone,

In the documentation webpage (http://ardrone-autonomy.readthedocs.org/en/latest/parameters.html) it is claimed that the parameters from a list can be changed.

Is the code to load the parameters in "launch/ardrone.launch" implemented?

I have been looking at "src/ardrone.driver" and I could not find the code, but I am not familiar with ROS parameters so maybe I am missing something obvious.

Thx in advance.

Josep

mani-monaj, 2015-11-06:
Most of the parameters are being read/set in [ardrone_sdk.cpp](https://github.com/AutonomyLab/ardrone_autonomy/blob/e1305b17efb18102d093206598a45cedd3ddb070/src/ardrone_sdk.cpp). 

Parameters that correspond to configurable entities in Ardrone SDK are set systematically by a set of `c` macros defined in [snippet_ros_to_ardrone_config.h](https://github.com/AutonomyLab/ardrone_autonomy/blob/e1305b17efb18102d093206598a45cedd3ddb070/include/ardrone_autonomy/snippet_ros_to_ardrone_config.h). `ardrone_sdk.cpp` [includes](https://github.com/AutonomyLab/ardrone_autonomy/blob/e1305b17efb18102d093206598a45cedd3ddb070/src/ardrone_sdk.cpp#L105) this file to set these parameters on initialization.

joseparnau, 2015-11-09:
Awesome!

Thank's!

mani-monaj, 2015-11-09:
@joseparnau You are welcome.

closed 2015-11-09 by mani-monaj