AutonomyLab/ardrone_autonomy #167: Makefile error

Sriherams, 2015-11-18:
Hi guys,

I'm a beginner in both ubuntu and ros field. Currently I'm using Ubuntu 15.04 and installed ros-indigo from the source by following the lines mentioned in the ros website. 

Moreover I tried to install the ardrone_autonomy by the steps mentioned in the docs, when I ran the code 'catkin_make' some error is popping and I'm not sure how to fix it. I mentioned my error as follows : 

Generating Lisp code from ardrone_autonomy/RecordEnable.srv
[ 78%] [ 79%] Generating C++ code from ardrone_autonomy/navdata_hdvideo_stream.msg
Generating C++ code from ardrone_autonomy/matrix33.msg
[ 80%] [ 81%] Generating C++ code from ardrone_autonomy/navdata_vision_of.msg
[ 82%] Generating C++ code from ardrone_autonomy/navdata_wind_speed.msg
[ 82%] Built target ardrone_autonomy_generate_messages_lisp
[ 82%] Generating C++ code from ardrone_autonomy/vector31.msg
Generating C++ code from ardrone_autonomy/navdata_references.msg
[ 83%] Generating C++ code from ardrone_autonomy/navdata_vision.msg
[ 84%] Generating C++ code from ardrone_autonomy/navdata_zimmu_3000.msg
[ 85%] Generating C++ code from ardrone_autonomy/navdata_gyros_offsets.msg
[ 86%] Generating C++ code from ardrone_autonomy/navdata_rc_references.msg
[ 87%] [ 87%] [ 88%] Generating C++ code from ardrone_autonomy/navdata_wifi.msg
Generating C++ code from ardrone_autonomy/vector21.msg
Generating C++ code from ardrone_autonomy/LedAnim.srv
[ 89%] Generating C++ code from ardrone_autonomy/CamSelect.srv
[ 90%] Generating C++ code from ardrone_autonomy/FlightAnim.srv
[ 91%] Generating C++ code from ardrone_autonomy/RecordEnable.srv
[ 91%] Built target ardrone_autonomy_generate_messages_cpp
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

mani-monaj, 2015-11-18:
Please provide the full output log, preferably by [pasting it into a gist](https://gist.github.com/) and providing the link here.

Sriherams, 2015-11-18:
Sorry my system got hang, so I have to reinstall everything. 
Does your ardrone_autonomy works with ros jade?

mani-monaj, 2015-11-18:
> Does your ardrone_autonomy works with ros jade?
 
Yes, but you need to compile from source. The binaries (debs) will be available very soon.

closed 2015-12-10 by Sriherams