AutonomyLab/ardrone_autonomy #170: Multiple Drones to the same PC

Geaper, 2015-12-12:
I am doing a C# Application that receives data from 2 different Drones. (I already configured their IP and i can ping them)
Drone1 -> 192.168.1.3
Drone2 -> 192.168.1.4

They both give me navigation_data from port 5554. I can read it but only one at a time. They both talk on the same port.

If I try to show the data on the labels I see drone1 data. If I disconnect the drone1 the data labels from drone2 start updating.

Is there a way to get data from the 2 devices using the same port?

mani-monaj, 2015-12-12:
Are you using `ardrone_autonomy` and `ROS`, or just the Parrot AR-Drone SDK?

Geaper, 2015-12-13:
An SDK built over the original.

https://github.com/tarazansafak/ArDroneControl/tree/master/AR.Drone-master

Geaper, 2015-12-13:
If I could change the ports from one drone it would be very easy, because i can change on the code.

kbogert, 2015-12-14:
This is the wrong place to ask this question, but multiple drone support works by having the client choose a random UDP port to open rather than 5553-5.  See here: https://github.com/AutonomyLab/ardronelib/commit/e30f9e0e58a4dbd7b831d487cd630914f01b568d

After a quick glace, it looks like in the ArDroneControl project if you change the corresponding:
```
 var udpClient = new UdpClient(NavdataPort)
```
to
```
 var udpClient = new UdpClient()
```

for the AT, Navdata, and Video ports it should work the same way.

Geaper, 2015-12-14:
I will try tomorrow. thanks for the help kbogert. love you. (no homo)

Geaper, 2015-12-14:
lol it worked kbogert. You are a genious. but if i connect both the label update is very slow and laggy even though the label timer update is 1ms on both.

I will give you my code. It's a Windows Forms app. 

Please change to var udpClient = new UdpClient(). 

I cannot upload again because my internet is very laggy :/

http://www.filedropper.com/tesesdknovo

mani-monaj, 2015-12-14:
@Geaper As @kbogert mentioned, this is not the right place to discuss this issue, the AR-Drone SDK that includes @kbogert's patch is [hosted here](https://github.com/AutonomyLab/ardronelib). You may want to manually diff that to your SDK to see all the patches we have applied so far (or equivalently browse that repo).  The ROS driver (this repo) is build against our modified version of the SDK and works  seamlessly with multiple drones (thanks to @kbogert) given a good network card.

closed 2015-12-14 by mani-monaj