AutonomyLab/ardrone_autonomy #171: Control Inputs

Sriherams, 2015-12-14:
Hi,
Can we able to change our control inputs from linear and angular velocities to actual control inputs provided by the AR Drone SDK 2.0 Developers guide? 
Since we are having some singularities while controlling the drone, is there a way
phi : Left/right angle 
theta : Front/back angle 
gaz : Vertical speed 
yaw : Angular speed

Thank You in advance...

mani-monaj, 2015-12-15:
Inputs are actually interpreted in the same way as you described. 

linear.x and linear.y control pitch and roll (angles) 
linear.z and angular.y control gaz and yaw (speed)

The conversion is done by the SDK/Drone by multiplying these values to corresponding `*max*` [parameter](http://ardrone-autonomy.readthedocs.org/en/latest/parameters.html). For example `linear.x` is multiplied by `euler_angle_max ` by the drone which results in a `linear.x * euler_angle_max` pitch.

Sriherams, 2015-12-21:
What about angular.x and angular.y?

mani-monaj, 2015-12-21:
Those degrees of freedom are not directly controllable, thus those inputs
are ignored.

Sriherams, 2015-12-22:
Thank You.

So in this repository the command velocity linear.x is directly linked to 'ardrone_at_set_progress_cmd' in the SDK - (i.e.) AT*PCMD = %d,%d,%d,%d,%d,%d (refer : AR Drone SDK Developer Guide)

Am I right?

mani-monaj, 2015-12-22:
Yes.

closed 2015-12-22 by mani-monaj