AutonomyLab/ardrone_autonomy #172: Too much memory used XXX mb > 40 mb, removing oldest flight ./flight

Riseley, 2015-12-16:
Successfully ran and manually controlled ardrone 2.0 on Ubuntu 14.04 with ROS indigo for a few flights.

I tried again after an hour later and things start to get strange.

No values found for `~cov/imu_la` in parameters, set all to zero
No values found for `~cov/imu_av` in parameters, set all to zero
No values found for `~cov/imu_or` in parameters, set all to zero
realtime navdata is off, odometry may be imprecise
Delete Profile -cea48d22
Deleting Application
...
...
Tag 28 is an unknown navdata option tag <-- Happens even before this issue
Tag 29 is an unknown navdata option tag <-- Happens even before this issue
...
...
Too much memory used 394 mb > 40 mb, removing oldest flight ./flight <-- kept looping the same thing, Ctrl-C could not interrupt and end the loop, must use terminate terminal

Tried resetting the drone and restarting.

I wonder what kind of issue is this.
Any help is greatly appreciated.

mani-monaj, 2015-12-16:
Please provide more information about your setup, like the firmware version and whether if you are running Ubuntu inside a virtual machine.

closed 2015-12-16 by mani-monaj

Riseley, 2015-12-16:
Sorry, but what do you mean by firmware? I have added information of ROS -> indigo and ardrone_autonomy to Nov 4 commit f7c66282c1f58b2cf8f33a30bb962242aa421c9b.

It seems like the too much memory problem always come out after I close a terminal of ardrone_autonomy. The next time I run ardrone_autonomy, no matter restarted or not, the problem will occur. A catkin_make seems to "solve" the problem. Even if nothing is changed in src.

mani-monaj, 2015-12-16:
That is strange. By firmware version, I mean the firmware version of your AR-Drone 2.0. You can either find that using Parrot's app or by looking at the output of the driver.

Riseley, 2015-12-18:
It is 2.4.8 by $telnet and $cat version.txt I tried to update it through AR.Freeflight on iOS and I believe both the app and the firmware is on the latest version.
I did some testings yesterday and this morning (That's why it took me so long to reply) and yes the problem is a consistent one. But I have to admit that my insufficient knowledge of ROS maybe the main cause. (Bad catkin declaration / ROS calls). I will look more into it by Monday.

P.S. Thank you for your work. I admire it a lot.

mani-monaj, 2015-12-22:
Great. Please update this issue if you find out more.