AutonomyLab/ardrone_autonomy #178: Stuck at "Getting ARDrone.Version"

achowdhery, 2016-02-26:
Hi

I am running code on ARDrone 2.0. I am able to test fine with a compiled ardrone SDK, but when I test ROS, the code gets stuck at "Getting ARDrone.Version." I have tried looking in to the issue. 
Any advice on how to debug this?

pulver22, 2016-03-03:
Can you paste all you got in the console?

mani-monaj, 2016-03-03:
1. Can you ping/telnet to your AR-Drone from the command line:

```
$ ping 192.168.1.1
$ telnet 192.168.1.1
```

2. Is your computer connected to multiple networks with the same IP range
as the drone (192.168.1.x)?

achowdhery, 2016-03-15:
I am able to do ping the drone - yes. Even control it with Linux SDK that comes with the AR drone. Just ROS doesn't work

mani-monaj, 2016-03-15:
@achowdhery How about `telnet`? Does it work? Could you please paste the output of `ifconfig` command on your machine? (Please remove any public IP information before pasting)

Hyllesen, 2016-04-07:
Having the same issue here on one Ubuntu machines. We can connect via. telnet and get the busybox prompt. We are able to connect from other machines, even Ubuntu/Xubuntu too.

mani-monaj, 2016-04-07:
> Update: We could suddenly connect after trying with telnet 192.168.1.1 and pinging ? Not sure what make the difference ?

Is this computer connected to multiple networks?

mani-monaj, 2016-04-14:
Closing this issue due to inactivity. Please feel free to re-open it in case you have new updates.

closed 2016-04-14 by mani-monaj

lilyospina, 2017-01-30:
Hi, I'm having the same problem!! I disconnected my wired connection and still get the message "Getting AR.Drone Version ...". I don't quite know what to do next.

mani-monaj, 2017-02-08:
@lilyospina What is the output of `ifconfig` on your machine? (Please remove any public IP information from the output before sharing it here). Can you `ping 192.168.1.1` successfully?

akarnati11, 2017-11-14:
Hi I am also getting this issue:
- I can ping 192.168.1.1 successfully
- I cannot telnet (terminal asks me for a password)
- This is the output of running ip:
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 00:1b:21:5b:ef:d1 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.10/24 brd 192.168.1.255 scope global eth1
       valid_lft forever preferred_lft forever
    inet6 fe80::21b:21ff:fe5b:efd1/64 scope link 
       valid_lft forever preferred_lft forever