AutonomyLab/ardrone_autonomy #180: package 'ardrone_autonomy' not found

usmanarif37, 2016-03-03:
HI
I am new to ROS and i am also learninig linux.
I have installed ROS Jade and and then installed ardrone_autonomy by first creating catkin workspace.
Installation was fine without errors.Now when i run this command:

rosrun ardrone_autonomy ardrone_driver
 
is get follwoing error:

package 'ardrone_autonomy' not found

 and in my src directory, i have ardrone_autonomy directory.. so how can my ros cannot find it.
confused

pulver22, 2016-03-03:
after moving to ~/catkin_ws (assuming this is your workspace), did you do 

`source devel/setub.bash`

?

usmanarif37, 2016-03-03:
yes i did that too. 
Now i deleted catkin workspaceand created new catkin workspace(i followed instructions on http://wiki.ros.org/catkin/Tutorials/create_a_workspace) and reinstalled my ardrone software. Still same problem

pulver22, 2016-03-03:
But why you didn't install from repo typing:
`sudo` apt-get install ros-indigo(or whatever)-ardrone-autonomy

?

usmanarif37, 2016-03-03:
I am new to this. The method is given by the AR drone autonomy. I don’t know any other method. If you could guide me, i can install it using repo.

pulver22, 2016-03-03:
Follow the instruction here:
[http://ardrone-autonomy.readthedocs.org/en/latest/installation.html](ARdrone autonomy reference)

I just tried and it worked! Install the software, connect your pc to the ardrone wifi hotspot and type in a terminal:

`rosrun ardone_autonomy ardrone_driver`

usmanarif37, 2016-03-03:
I have been using the same link to install.After installation, in the next step of "Usage", there is command given:
$ rosrun ardrone_autonomy ardrone_driver 
This command is not working. It gives the error:
$ package 'ardrone_autonomy' not found
 This is what the problem is :(

pulver22, 2016-03-03:
Do you have in your .bashrc file something like this:

`#ROS variables
source /opt/ros/indigo/setup.bash
source ~/catkin_ws/devel/setup.bash
source ~/catkin_ws2/devel/setup.bash

ROS_PACKAGE_PATH=/home/pulver/catkin_ws:$ROS_PACKAGE_PATH
`
I think your problem is more related to a wrong installation of ROS

usmanarif37, 2016-03-03:
It contains
1. source /opt/ros/indigo/setup.bash

but doesn’t contain
1.source ~/catkin_ws/devel/setup.bash
2.source ~/catkin_ws2/devel/setup.bash

also the ros package path is:
/opt/ros/jade/share:/opt/ros/jade/stacks"
i have attached my bashrc file
[document.txt](https://github.com/AutonomyLab/ardrone_autonomy/files/156996/document.txt)

pulver22, 2016-03-03:
In your .bashrc file I found

`source /opt/ros/jade/setup.bash`

This means you have ROS Jade. Therefore, when you try to install ardrone_autonomy from repo, you should type:
`sudo apt-get install ros-jade-ardrone-autonomy`

I hope this helps you.

usmanarif37, 2016-03-03:
Yeah... 
GOD bless you.
This worked. Thank you so much. :) :100:  :+1:

pulver22, 2016-03-03:
Happy for helping you ;)

closed 2016-03-03 by usmanarif37