AutonomyLab/ardrone_autonomy #181: ar_pose + ardrone_autonomy

pulver22, 2016-03-15:
Hi all,

I'm trying to perform marker detection using the ar_pose package. 
Everything works fine using the package alone, either wih the webcam or an usb camera, getting the video stream with gscam.

I tried to write a launch file that launch ar_pose and ardrone_autonomy simultaneously, in order to use the frontcam of the quadcopter as camera for the acquisition. Here is the launch file:

`<launch>
    
    <arg name="ip" default="192.168.1.1"/>

    
    <arg name="freq" default="8"/>
    <node name="ardrone_driver" pkg="ardrone_autonomy" type="ardrone_driver" output="screen" clear_params="true" args="-ip $(arg ip)">
        <param name="outdoor" value="0"/>
        <param name="max_bitrate" value="4000"/>
        <param name="bitrate" value="4000"/>
        <param name="navdata_demo" value="0"/>
        <param name="flight_without_shell" value="0"/>
        <param name="altitude_max" value="4000"/>
        <param name="altitude_min" value="50"/>
        <param name="euler_angle_max" value="0.21"/>
        <param name="control_vz_max" value="700"/>
        <param name="control_yaw" value="1.75"/>
        <param name="detect_type" value="10"/>
        <param name="enemy_colors" value="3"/>
        <param name="detections_select_h" value="32"/>
        <param name="detections_select_v_hsync" value="128"/>
        <param name="enemy_without_shell" value="0"/>
        <param name="ultrasound_freq" value="$(arg freq)"/>
        <param name="realtime_navdata" value="true"/>
        <param name="realtime_video" value="true"/>
        
        <rosparam param="cov/imu_la">[0.1, 0.0, 0.0, 0.0, 0.1, 0.0, 0.0, 0.0, 0.1]</rosparam>
        <rosparam param="cov/imu_av">[1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]</rosparam>
        <rosparam param="cov/imu_or">[1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 100000.0]</rosparam>
    </node>

    
    
    
    

    <node pkg="tf" type="static_transform_publisher" name="cam_to_drone" args="1 0 0 0 0 0 world ardrone_base_frontcam 10"/>

    
    <node ns="camera" pkg="image_proc" type="image_proc" name="image_proc"/>

    
    

    <node name="ar_pose" pkg="ar_pose" type="ar_single" respawn="false" output="screen">
        <param name="marker_pattern" type="string" value="$(find ar_pose)/data/4x4/4x4_1.patt"/>
        <param name="marker_width" type="double" value="152.4"/>
        <param name="marker_center_x" type="double" value="0.0"/>
        <param name="marker_center_y" type="double" value="0.0"/>
        <param name="threshold" type="int" value="100"/>
        <param name="use_history" type="bool" value="true"/>
    </node>

    
    

</launch>
`

The problem is that in this way no marker are detected. 
If I uncomment the gscam part of the launch file, I am able to perceive marker only from device=/dev/video0 that is the webcam.

Have you ever tried to perform this operation?

I think one solution could be modified the frame of ardrone in the source, changing "ardrone_base_frontcam" in "camera" but I'm not sure. I think so because it seems that, as ar_pose is written, marker can be attached only to "camera" frame (of course another attempt woulbe changing the ar_pose code).

in the hypothesis I don't want to change the code (to not recompile it) is there a way to do it inside the launch file?

mani-monaj, 2016-03-15:
I think you need a proper [remapping](http://wiki.ros.org/roslaunch/XML/remap) from `ardrone/image_raw` to `camera/image_raw` (or whatever topic `arpose` requires as an input). If you paste the result of `rostopic list` on your system, I can suggest the correct `<remap>` tag.

pulver22, 2016-03-16:
Here's the output of `rostopic list`:

```
/ar_pose_marker
/ardrone/bottom/camera_info
/ardrone/bottom/image_raw
/ardrone/bottom/image_raw/compressed
/ardrone/bottom/image_raw/compressed/parameter_descriptions
/ardrone/bottom/image_raw/compressed/parameter_updates
/ardrone/bottom/image_raw/compressedDepth
/ardrone/bottom/image_raw/compressedDepth/parameter_descriptions
/ardrone/bottom/image_raw/compressedDepth/parameter_updates
/ardrone/bottom/image_raw/theora
/ardrone/bottom/image_raw/theora/parameter_descriptions
/ardrone/bottom/image_raw/theora/parameter_updates
/ardrone/camera_info
/ardrone/front/camera_info
/ardrone/front/image_raw
/ardrone/front/image_raw/compressed
/ardrone/front/image_raw/compressed/parameter_descriptions
/ardrone/front/image_raw/compressed/parameter_updates
/ardrone/front/image_raw/compressedDepth
/ardrone/front/image_raw/compressedDepth/parameter_descriptions
/ardrone/front/image_raw/compressedDepth/parameter_updates
/ardrone/front/image_raw/theora
/ardrone/front/image_raw/theora/parameter_descriptions
/ardrone/front/image_raw/theora/parameter_updates
/ardrone/image_raw
/ardrone/image_raw/compressed
/ardrone/image_raw/compressed/parameter_descriptions
/ardrone/image_raw/compressed/parameter_updates
/ardrone/image_raw/compressedDepth
/ardrone/image_raw/compressedDepth/parameter_descriptions
/ardrone/image_raw/compressedDepth/parameter_updates
/ardrone/image_raw/theora
/ardrone/image_raw/theora/parameter_descriptions
/ardrone/image_raw/theora/parameter_updates
/ardrone/imu
/ardrone/land
/ardrone/mag
/ardrone/navdata
/ardrone/odometry
/ardrone/reset
/ardrone/takeoff
/camera/camera_info
/camera/image_color
/camera/image_color/compressed
/camera/image_color/compressed/parameter_descriptions
/camera/image_color/compressed/parameter_updates
/camera/image_color/compressedDepth
/camera/image_color/compressedDepth/parameter_descriptions
/camera/image_color/compressedDepth/parameter_updates
/camera/image_color/theora
/camera/image_color/theora/parameter_descriptions
/camera/image_color/theora/parameter_updates
/camera/image_mono
/camera/image_mono/compressed
/camera/image_mono/compressed/parameter_descriptions
/camera/image_mono/compressed/parameter_updates
/camera/image_mono/compressedDepth
/camera/image_mono/compressedDepth/parameter_descriptions
/camera/image_mono/compressedDepth/parameter_updates
/camera/image_mono/theora
/camera/image_mono/theora/parameter_descriptions
/camera/image_mono/theora/parameter_updates
/camera/image_proc_debayer/parameter_descriptions
/camera/image_proc_debayer/parameter_updates
/camera/image_proc_rectify_color/parameter_descriptions
/camera/image_proc_rectify_color/parameter_updates
/camera/image_proc_rectify_mono/parameter_descriptions
/camera/image_proc_rectify_mono/parameter_updates
/camera/image_rect
/camera/image_rect/compressed
/camera/image_rect/compressed/parameter_descriptions
/camera/image_rect/compressed/parameter_updates
/camera/image_rect/compressedDepth
/camera/image_rect/compressedDepth/parameter_descriptions
/camera/image_rect/compressedDepth/parameter_updates
/camera/image_rect/theora
/camera/image_rect/theora/parameter_descriptions
/camera/image_rect/theora/parameter_updates
/camera/image_rect_color
/camera/image_rect_color/compressed
/camera/image_rect_color/compressed/parameter_descriptions
/camera/image_rect_color/compressed/parameter_updates
/camera/image_rect_color/compressedDepth
/camera/image_rect_color/compressedDepth/parameter_descriptions
/camera/image_rect_color/compressedDepth/parameter_updates
/camera/image_rect_color/theora
/camera/image_rect_color/theora/parameter_descriptions
/camera/image_rect_color/theora/parameter_updates
/cmd_vel
/rosout
/rosout_agg
/tf
/visualization_marker
```

mani-monaj, 2016-03-16:
Try this:

```xml
<node name="ardrone_driver" pkg="ardrone_autonomy" type="ardrone_driver" output="screen" clear_params="true" args="-ip $(arg ip)">
    ...
    <remap from="ardrone/front/image_raw" to="camera/image_color" />
    ...
</node>
```

Also, you probably do not need the `static_transform_publisher` node since the driver already [publishes](http://ardrone-autonomy.readthedocs.org/en/latest/frames.html) the proper transform from the camera frame to the `base_link`.

pulver22, 2016-03-17:
The problems are partially solved. 

Now I can detect the marker but its frame is fixed to the one of the drone camera [link](http://imgur.com/m9CraZV) instead moving around as it should do [link](http://imgur.com/NzyQBpe).

mani-monaj, 2016-03-17:
Can you please visualize all TF frames for both of the cases and share the results (`frame.pdf`):

```bash
$ rosrun tf view_frames
```

pulver22, 2016-03-18:
I solved the problem, I'll try to clarify the situation.

I discovered that the marker was attached to the camera frame because I was visualizing the Image received on the topic, without its depth. 

Trying to visualize the `Camera` instead of the `Image` in Rviz I got the following error:

`Error: CameraInfo/P resulted in an invalid position calculation`

Investigating the error above I discovered that the camera need to be calibrated to perceive the depth of the image broadcasted on the topic.

After doing so I was able to see the marker moving around the camera frame!!

Thank you @mani-monaj for your support!

mani-monaj, 2016-03-18:
@pulver22 You are welcome. Good luck!

closed 2016-03-18 by mani-monaj