AutonomyLab/ardrone_autonomy #185: Navdata altd field incorrect after a couple of minutes

jay3ss, 2016-04-09:
I've noticed that after a few minutes of flying, sometimes the atld field of the /ardrone/navdata topic is incorrect. For example, it will show that the drone is at an altitude of approximately 2 to 3 meters when in reality it's at an altitude of less than 0.5 meters. I've also seen it show negative altitudes as well when it was close to the floor. Could this be a hardware issue, driver issue, or something else? I'm using the AR.Drone 2.0, ardrone_autonomy for Indigo, and Ubuntu 14.04.

mani-monaj, 2016-04-11:
I've never seen this issue myself. However I should mention I was not relying on `altd` for long flights during any of my experiments with AR-Drone 2.0 . It would be useful if you can test with another AR-Drone 2.0 to see if that also shows the same behaviour. Parrot's original SDK comes with an example GUI application to pilot/monitor an AR-Drone. You could also try compiling/running that application to check if you see the same behavior.

jay3ss, 2016-04-11:
I ordered another Ar.Drone 2.0 for my own personal use and I'll confirm this issue with that. I also hadn't been using `altd` for extended periods of time until recently so I can't say whether or not this is a new issue.