AutonomyLab/ardrone_autonomy #188: compile from source - problem

jonyballerino, 2016-04-18:
Hi, I'm trying to compile from source ardrone_autonomy, following the instruction on documentation.
It's all ok, but when I run `catkin_make`, process try to built target ardrone_autonomy_generate_messages and it says to me that it's impossible to connect to github.com.

```
myname ~/catkin_ws # catkin_make
Base path: /home/myname/catkin_ws
Source space: /home/myname/catkin_ws/src
Build space: /home/myname/catkin_ws/build
Devel space: /home/myname/catkin_ws/devel
Install space: /home/myname/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/myname/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/myname/catkin_ws/build"
####
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_euler_angles
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_vision_raw
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_vision_perf
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_vector21
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_matrix33
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_vision_detect
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_zimmu_3000
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_LedAnim
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_time
[  0%] [  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_raw_measures
Built target _ardrone_autonomy_generate_messages_check_deps_navdata_vision
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_video_stream
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_trims
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_gyros_offsets
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_watchdog
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_wind_speed
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_altitude
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_references
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_phys_measures
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_vector31
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_vision_of
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_rc_references
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_magneto
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_kalman_pressure
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_hdvideo_stream
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_demo
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_FlightAnim
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_games
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_pwm
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_Navdata
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_wifi
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_adc_data_frame
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_pressure_raw
[  0%] Built target _ardrone_autonomy_generate_messages_check_deps_CamSelect
[  0%] [  0%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_trackers_send
Built target geometry_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] [  0%] Built target std_msgs_generate_messages_lisp
Built target geometry_msgs_generate_messages_lisp
[  1%] [  1%] Built target _ardrone_autonomy_generate_messages_check_deps_RecordEnable
Performing download step (git clone) for 'ardronelib'
Cloning into 'ardronelib'...
[ 30%] Built target ardrone_autonomy_generate_messages_cpp
[ 60%] Built target ardrone_autonomy_generate_messages_lisp
[ 90%] Built target ardrone_autonomy_generate_messages_py
[ 90%] Built target ardrone_autonomy_gencpp
[ 90%] Built target ardrone_autonomy_generate_messages
fatal: unable to connect to github.com:
github.com[0: 192.30.252.128]: errno=Connessione scaduta

Cloning into 'ardronelib'...
fatal: unable to connect to github.com:
github.com[0: 192.30.252.128]: errno=Connessione scaduta

Cloning into 'ardronelib'...
fatal: unable to connect to github.com:
github.com[0: 192.30.252.120]: errno=Connessione scaduta

-- Had to git clone more than once:
          3 times.
CMake Error at /home/myname/catkin_ws/devel/tmp/ardronelib-gitclone.cmake:40 (message):
  Failed to clone repository: 'git://github.com/AutonomyLab/ardronelib.git'

make[2]: *** [/home/myname/catkin_ws/devel/src/ardronelib-stamp/ardronelib-download] Errore 1
make[1]: *** [ardrone_autonomy/CMakeFiles/ardronelib.dir/all] Errore 2
make: *** [all] Errore 2
```

Can anybody help me?

pulver22, 2016-04-19:
I had the same issue and in my case it was related to the network I was using.
I'm suggesting you to download it from the ros repo :+1:

closed 2016-04-19 by jonyballerino

mani-monaj, 2016-04-19:
@jonyballerino Are you connected to the internet when compiling from source?

ramanathanrm95, 2017-01-18:
@mani-monaj I am facing the same problem. I am connected to the internet when compiling. Is it a timeout problem or what?
If so, how can I rectify it?

mani-monaj, 2017-02-08:
@ramanathanrm95 Could you please provide more information and re-open this issue?

ramanathanrm95, 2017-02-08:
Thanks for your reply...I solved the issue...there was some problem with my
Wi-Fi network

mani-monaj, 2017-02-08:
@ramanathanrm95 Thanks for the update.

Pravin854, 2019-01-22:
In CMakeLists.txt file change git://github.com/AutonomyLab/ardronelib.git to https://github.com/AutonomyLab/ardronelib.git

DediuAndrei21, 2023-02-25:
> In CMakeLists.txt file change git://github.com/AutonomyLab/ardronelib.git to https://github.com/AutonomyLab/ardronelib.git

Can you please tell me the path to this folder?

yacinmkh21, 2024-04-17:
> > In CMakeLists.txt file change git://github.com/AutonomyLab/ardronelib.git to https://github.com/AutonomyLab/ardronelib.git
> 
> Can you please tell me the path to this folder? I cand find a CMakeList that contain the git link for ardronelib.

did you find the solution?

yacinmkh21, 2024-04-17:
> > In CMakeLists.txt file change git://github.com/AutonomyLab/ardronelib.git to https://github.com/AutonomyLab/ardronelib.git
> 
> Can you please tell me the path to this folder? I cand find a CMakeList that contain the git link for ardronelib.

how to solve the probleme

yacinmkh21, 2024-04-17:
> > In CMakeLists.txt file change git://github.com/AutonomyLab/ardronelib.git to https://github.com/AutonomyLab/ardronelib.git
> 
> Can you please tell me the path to this folder? I cand find a CMakeList that contain the git link for ardronelib.

how to solve the probleme,please