AutonomyLab/ardrone_autonomy #190: navdata_gps reference system

SimonKinds, 2016-04-21:
Hey!

I'm using the gps-waypoint branch and am able to successfully read the navdata_gps topic. 
When defining a GeoPoint the WGS 84 reference system is used, but I'm wondering if this is also true for the raw latitude / longitude data read from navdata_gps.

Another thing I'm wondering is the purpose of the lat0, long0, lat_fused and long_fused variables.

SimonKinds, 2016-04-21:
@lesire, do you know anything about this? Asking because it seems you were the one that found the header specification.

lesire, 2016-04-21:
IIRC data from the GPS receiver are in WGS84 lat/long format.
Know nothing about {lat,long}{0,_fused}

SimonKinds, 2016-04-21:
Thank you for the answer, much appreciated.

SimonKinds, 2016-04-21:
When handling a new GPS-waypoint there are some magic happening, I'm wondering if you (or someone else) might be able to clear some of them up.
(https://github.com/AutonomyLab/ardrone_autonomy/blob/gps-waypoint/src/teleop_twist.cpp#L175)

The lat/long are multiplied by 1.0e7, for no apparent reason. Is this maybe some kind of conversion between WGS84 and another reference system. 
Furthermore, the altitude is multiplied by 1000 to seemingly convert it to mm, does this mean that the elevation values given from navdata_gps are in mm?

There are also some constants given when creating param_str which I'm bewildered about, but I have a feeling those are not important in the context of reference systems.

mani-monaj, 2016-04-22:
> When handling a new GPS-waypoint there are some magic happening, I'm wondering if you (or someone else) might be able to clear some of them up.
(https://github.com/AutonomyLab/ardrone_autonomy/blob/gps-waypoint/src/teleop_twist.cpp#L175)
> The lat/long are multiplied by 1.0e7, for no apparent reason. Is this maybe some kind of conversion between WGS84 and another reference system. 

I found out the format of the data sent to AR-Drone in `flying_camera_mode` through reverse engineering (packet sniffing). The conversations are just to make all the values into `int32_t`s with enough precision. I don't believe they have anything to do with changing the coordinate system.

> Furthermore, the altitude is multiplied by 1000 to seemingly convert it to mm, does this mean that the elevation values given from navdata_gps are in mm?

I can not quite recall what the unit was. However it should be trivial to find that out by looking at its change when the drone is at certain pre-known reference heights.

> There are also some constants given when creating param_str which I'm bewildered about, but I have a feeling those are not important in the context of reference systems.

You are right.

closed 2016-04-22 by SimonKinds