AutonomyLab/ardrone_autonomy #191: All navdatas are always at 0

Gingerbreadz, 2016-05-29:
Hey,
I'm currently working with an ardrone 2.0 with ROS. The problem is, navdatas are always at the same value : 0. I've successfully got the drone to move around by sending ROS messages, but when I listen the navdata topic, all field are at 0. 
A previous team has been working with this drone and when I first started the ardrone_autonomy driver, the displayed firmware was "0.0.0" and the battery percentage was always at 0. I tried to control the drone with the "AR Free Flight" android app, but it stays stuck on "loading". 
So I flashed the 2.4.8 firmware, the ROS driver now detects the correct firmware version, but the navdatas and battery percentage are still at 0 and still unable to fly with the android app.
I tried many older firmwares with no success, and even tried to downgrade manually to then let the android app update the drone to the latest firmware but it didn't solve anything. I also tried to press the hardware reset button with no success.

I couldn't check if this were some hardware-related issue, or if it were due to modifications to the drone system, from the previous users, that weren't erased when I flashed firmwares. 
If anyone has experienced this or have any clue about the problem, I'd be thankful if you help.

mani-monaj, 2016-05-30:
> So I flashed the 2.4.8 firmware, the ROS driver now detects the correct firmware version, but the navdatas and battery percentage are still at 0 and still unable to fly with the android app.

Since you can not control your AR-Drone with the official Android App, there seems to be something wrong on the fimrware (or hardware) side of the drone. I suggest that you ask this question on Parrot [support](https://community.parrot.com/t5/AR-Drone-2/bd-p/ARDrone_EN) or [developer](http://forum.developer.parrot.com/) forum.

mani-monaj, 2016-06-18:
Please consider re-opening this issue if you have further updates to share.

closed 2016-06-18 by mani-monaj