AutonomyLab/ardrone_autonomy #193: Multiple AR.Drone 2.0 near each other results in unexpected behaviour

Geaper, 2016-06-09:
I am using 2 AR.Drones together and one of them starts going up and one down for example. I though it's was the sonar frequency and I telnet into each AR drone and changed vi /data/config.ini - > ultrasound_frequency so they are different and both have ultrasound_watchdog = 3 and one of them still going up into the ceiling when they are near each other.

mani-monaj, 2016-06-18:
This behaviour is most likely due to sonar interference. Try passing the `ultrasound_freq` parameter to the driver. As far as I remember, you need to set it to `7` for one drone and `8` for the other one. More info:

-http://ardrone-autonomy.readthedocs.io/en/latest/parameters.html
- https://github.com/AutonomyLab/ardrone_autonomy/wiki/AR-Drone-Parameters

Geaper, 2016-06-18:
Yeah I already had telnet and changed to 7 and 8 in vi /data/config.ini.

mani-monaj, 2016-06-18:
The driver might be resetting them when you run it.

Geaper, 2016-06-18:
I will try thx. I will reply this week when I test.

Geaper, 2016-06-18:
btw, do you know the minimum and max values of ultrasom_frequency?

mani-monaj, 2016-06-18:
It's either `7` or `8`

Geaper, 2016-06-18:
Do you have any idea what's watchdog = 3 is for? Should I leave it both at 3? Thanks

mani-monaj, 2016-06-18:
No unfortunately

Geaper, 2016-06-18:
Ok thx. One more question. If I have 4 drones, I can only use 2 different sonar frequencies? That will cause interfere between 2-2.

mani-monaj, 2016-06-18:
That is true. However you can spread them apart such as the neighbouring drones use different frequencies.

Geaper, 2016-06-20:
Seems to be better now.

But I cant control very well because every 2min or so they lose connections and reconnect again. I'm running tum_ardrone and if It starts reconnecting it will hold the last command and if it's movement and it starts reconnecting the drone can go into a wall.

Any idea?

`Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
[... 77 lines cut ...]
[ INFO] [1466434928.729993592]:   SEND: CAT_APPLI/navdata_options = 268435456.000000 (DEFAULT = 65537.000000)
[ INFO] [1466434928.730056748]:   SEND: CAT_SESSION/video_codec = 129.000000 (DEFAULT = 32.000000)
[ INFO] [1466434928.730080918]:   SEND: CAT_SESSION/max_bitrate = 4000.000000 (DEFAULT = 1000.000000)
[ INFO] [1466434928.755497237]: Successfully connected to 'My ARDrone' (AR-Drone 2.0 - Firmware: 2.4.8) - Battery(%): 52
[ INFO] [1466434928.755536320]: Navdata Publish Settings:
[ INFO] [1466434928.755549023]:     Legacy Navdata Mode: On
[ INFO] [1466434928.755557908]:     ROS Loop Rate: 200 Hz
[ INFO] [1466434928.755566412]:     Realtime Navdata Publish: On
[ INFO] [1466434928.755576643]:     Realtime Video Publish: On
[ INFO] [1466434928.755587409]:     Drone Navdata Send Speed: 200Hz (navdata_demo=0)
[ INFO] [1466434928.767483810]:   SEND: CAT_APPLI/navdata_options = 268435456.000000 (DEFAULT = 65537.000000)
[ INFO] [1466434928.767542948]:   SEND: CAT_SESSION/video_codec = 129.000000 (DEFAULT = 32.000000)
[ INFO] [1466434928.767561523]:   SEND: CAT_SESSION/max_bitrate = 4000.000000 (DEFAULT = 1000.000000)
[ INFO] [1466434928.791377365]: Successfully connected to 'My ARDrone' (AR-Drone 2.0 - Firmware: 2.4.8) - Battery(%): 90
[ INFO] [1466434928.791482851]: Navdata Publish Settings:
[ INFO] [1466434928.791504667]:     Legacy Navdata Mode: On
[ INFO] [1466434928.791518363]:     ROS Loop Rate: 200 Hz
[ INFO] [1466434928.791538090]:     Realtime Navdata Publish: On
[ INFO] [1466434928.791552704]:     Realtime Video Publish: On
[ INFO] [1466434928.791570312]:     Drone Navdata Send Speed: 200Hz (navdata_demo=0)
Timeout when reading navdatas - resending a navdata request on port 5554
Reconnecting ... OK
Reconnecting ... Reconnecting ... OK
OK
`

mani-monaj, 2016-06-20:
It's probably due to wifi interference. How does your setup look like? One PC per drone or multiple drones connected to a router?

Geaper, 2016-06-20:
I tried one pc per drone and one pc with a router and both do the same. Maybe I need to lower the nao data frequency?  I set it to 200hz

Geaper, 2016-06-20:
Navdata*

kbogert, 2016-06-21:
Are you placing them on wifi  channels as far away from each other as you can? (think 1 and 11)  The drones are very sensitive to wifi interference thanks to transferring video over TCP (don't ask).  In comparison the navdata is a tiny UDP packet, even at 200hz it's not that big of a problem, and lowering it may cause trouble with tum_ardrone. 

In your ardrone.launch file, remove the bitrate config option and set max_bitrate down to 1000.

Another thing you might try is turning on dynamic bitrate, edit line 93 of src/ardrone_sdk.cpp to be:

ardrone_application_default_config.bitrate_ctrl_mode = VBC_MODE_DYNAMIC;

Geaper, 2016-06-21:
Hello.

I have the wifi router in a complete different channel than the nearby wifi.

I did what you said and it's even worse. 

> Timeout when reading navdatas - resending a navdata request on port 5554
[ INFO] [1466521870.269624249]:   SEND: CAT_APPLI/navdata_options = 268435456.000000 (DEFAULT = 65537.000000)
[ INFO] [1466521870.269779522]:   SEND: CAT_SESSION/video_codec = 129.000000 (DEFAULT = 32.000000)
[ INFO] [1466521870.269864253]:   SEND: CAT_SESSION/max_bitrate = 4000.000000 (DEFAULT = 1000.000000)
Reconnecting ... Reconnecting ... OK
Reconnecting ... [ INFO] [1466521870.299491395]: Successfully connected to 'My ARDrone' (AR-Drone 2.0 - Firmware: 2.4.8) - Battery(%): 7
[ INFO] [1466521870.299548491]: Navdata Publish Settings:
[ INFO] [1466521870.299575287]:     Legacy Navdata Mode: On
[ INFO] [1466521870.299614734]:     ROS Loop Rate: 200 Hz
[ INFO] [1466521870.299636472]:     Realtime Navdata Publish: On
[ INFO] [1466521870.299657950]:     Realtime Video Publish: On
[ INFO] [1466521870.299685742]:     Drone Navdata Send Speed: 200Hz (navdata_demo=0)
[ INFO] [1466521870.601530334]:   SEND: CAT_APPLI/navdata_options = 268435456.000000 (DEFAULT = 65537.000000)
[ INFO] [1466521870.601651589]:   SEND: CAT_SESSION/video_codec = 129.000000 (DEFAULT = 32.000000)
[ INFO] [1466521870.601708982]:   SEND: CAT_SESSION/max_bitrate = 4000.000000 (DEFAULT = 1000.000000)
[ INFO] [1466521870.627213873]: Successfully connected to 'My ARDrone' (AR-Drone 2.0 - Firmware: 2.4.8) - Battery(%): 48
[ INFO] [1466521870.627263118]: Navdata Publish Settings:
[ INFO] [1466521870.627293112]:     Legacy Navdata Mode: On
[ INFO] [1466521870.627320910]:     ROS Loop Rate: 200 Hz
[ INFO] [1466521870.627346778]:     Realtime Navdata Publish: On
[ INFO] [1466521870.627370919]:     Realtime Video Publish: On
[ INFO] [1466521870.627397573]:     Drone Navdata Send Speed: 200Hz (navdata_demo=0)
Timeout when reading navdatas - resending a navdata request on port 5554
Timeout when reading navdatas - resending a navdata request on port 5554
Reconnecting ... Reconnecting ... OK
Timeout when reading navdatas - resending a navdata request on port 5554
OK
OK
Reconnecting ... OK
OK
Reconnecting ... OK
Timeout when reading navdatas - resending a navdata request on port 5554
Reconnecting ... OK
Reconnecting ... OK
Reconnecting ... OK

Geaper, 2016-06-21:
<launch>
<group ns="drone1">
        <node name="drone1" pkg="ardrone_autonomy" type="ardrone_driver" args="-ip 192.168.1.10" output="screen" clear_params="true"/>
	    <param name="outdoor" value="0" />
            <param name="max_bitrate" value="1000" />
	    <param name="navdata_demo" value="0" />
            <param name="realtime_navdata" value="true" />
            <param name="realtime_video" value="true" />
	    <param name="flight_without_shell" value="0" />
	    <param name="altitude_max" value="3000" />
            <param name="ultrasound_freq" value="7" />
        <param name="altitude_min" value="50" />
        <param name="euler_angle_max" value="0.21" />
        <param name="control_vz_max" value="700" />
	    <param name="control_yaw" value="1.75" />
	    <param name="detect_type" value="10" />
	    <param name="enemy_colors" value="3" />
	    <param name="detections_select_h" value="32" />
	    <param name="detections_select_v_hsync" value="128" />
	    <param name="enemy_without_shell" value="0" />
	    <param name="do_imu_caliberation" value="false" />
	    <param name="tf_prefix" value="mydrone" />
        
        <rosparam param="cov/imu_la">[0.1, 0.0, 0.0, 0.0, 0.1, 0.0, 0.0, 0.0, 0.1]</rosparam>
        <rosparam param="cov/imu_av">[1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]</rosparam>
        <rosparam param="cov/imu_or">[1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 100000.0]</rosparam>

</group>
	
<group ns="drone2">
        <node name="drone2" pkg="ardrone_autonomy" type="ardrone_driver" args="-ip 192.168.1.20" output="screen" clear_params="true"/>
	    <param name="outdoor" value="0" />
            <param name="max_bitrate" value="1000" />
	    <param name="navdata_demo" value="0" />
            <param name="realtime_navdata" value="True" />
            <param name="realtime_video" value="True" />
	    <param name="flight_without_shell" value="0" />
	    <param name="altitude_max" value="3000" />
            <param name="ultrasound_freq" value="8" />
        <param name="altitude_min" value="50" />
        <param name="euler_angle_max" value="0.21" />
        <param name="control_vz_max" value="700" />
	    <param name="control_yaw" value="1.75" />
	    <param name="detect_type" value="10" />
	    <param name="enemy_colors" value="3" />
	    <param name="detections_select_h" value="32" />
	    <param name="detections_select_v_hsync" value="128" />
	    <param name="enemy_without_shell" value="0" />
	    <param name="do_imu_caliberation" value="false" />
	    <param name="tf_prefix" value="mydrone" />
        
        <rosparam param="cov/imu_la">[0.1, 0.0, 0.0, 0.0, 0.1, 0.0, 0.0, 0.0, 0.1]</rosparam>
        <rosparam param="cov/imu_av">[1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]</rosparam>
        <rosparam param="cov/imu_or">[1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 100000.0]</rosparam>
</group>
</launch>

Geaper, 2016-06-21:
`<launch>
<group ns="drone1">
        <node name="drone1" pkg="ardrone_autonomy" type="ardrone_driver" args="-ip 192.168.1.10" output="screen" clear_params="true"/>
	    <param name="outdoor" value="0" />
            <param name="max_bitrate" value="1000" />
	    <param name="navdata_demo" value="0" />
            <param name="realtime_navdata" value="true" />
            <param name="realtime_video" value="true" />
	    <param name="flight_without_shell" value="0" />
	    <param name="altitude_max" value="3000" />
            <param name="ultrasound_freq" value="7" />
        <param name="altitude_min" value="50" />
        <param name="euler_angle_max" value="0.21" />
        <param name="control_vz_max" value="700" />
	    <param name="control_yaw" value="1.75" />
	    <param name="detect_type" value="10" />
	    <param name="enemy_colors" value="3" />
	    <param name="detections_select_h" value="32" />
	    <param name="detections_select_v_hsync" value="128" />
	    <param name="enemy_without_shell" value="0" />
	    <param name="do_imu_caliberation" value="false" />
	    <param name="tf_prefix" value="mydrone" />
        
        <rosparam param="cov/imu_la">[0.1, 0.0, 0.0, 0.0, 0.1, 0.0, 0.0, 0.0, 0.1]</rosparam>
        <rosparam param="cov/imu_av">[1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]</rosparam>
        <rosparam param="cov/imu_or">[1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 100000.0]</rosparam>

</group>
	
<group ns="drone2">
        <node name="drone2" pkg="ardrone_autonomy" type="ardrone_driver" args="-ip 192.168.1.20" output="screen" clear_params="true"/>
	    <param name="outdoor" value="0" />
            <param name="max_bitrate" value="1000" />
	    <param name="navdata_demo" value="0" />
            <param name="realtime_navdata" value="True" />
            <param name="realtime_video" value="True" />
	    <param name="flight_without_shell" value="0" />
	    <param name="altitude_max" value="3000" />
            <param name="ultrasound_freq" value="8" />
        <param name="altitude_min" value="50" />
        <param name="euler_angle_max" value="0.21" />
        <param name="control_vz_max" value="700" />
	    <param name="control_yaw" value="1.75" />
	    <param name="detect_type" value="10" />
	    <param name="enemy_colors" value="3" />
	    <param name="detections_select_h" value="32" />
	    <param name="detections_select_v_hsync" value="128" />
	    <param name="enemy_without_shell" value="0" />
	    <param name="do_imu_caliberation" value="false" />
	    <param name="tf_prefix" value="mydrone" />
        
        <rosparam param="cov/imu_la">[0.1, 0.0, 0.0, 0.0, 0.1, 0.0, 0.0, 0.0, 0.1]</rosparam>
        <rosparam param="cov/imu_av">[1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]</rosparam>
        <rosparam param="cov/imu_or">[1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 100000.0]</rosparam>
</group>
</launch>

`

kbogert, 2016-06-21:
By nearby wifi you mean the second wifi adapter for drone #2, or another access point nearby?  What channels are you using?

I've never gotten more than 3 drones flying at the same time, and then I only did so by moving them to an area with no wifi AP's nearby (in a park) with two laptops and an access point set to channels 1, 6, and 11.

Also, the log you sent still says the max_bitrate is 4000, is this when you were trying with dynamic bitrate enabled?

mani-monaj, 2016-06-23:
I agree with @kbogert 's suggestions. We've had a lot of problems with Wifi interference among drones previously. Also please note that some access points perform poorly under load, you can for example check the experiments section of [this work](http://ieeexplore.ieee.org/xpl/articleDetails.jsp?reload=true&arnumber=6816845) by our group to see how bad the situation can become in a similar condition.

sagerd, 2016-09-15:
I'm having this exact same issue.  Did anyone ever resolve this problem or find a way around it?  When running 2 drones I can't even issue commands after takeoff because of this.

mani-monaj, 2016-10-14:
@sagerd Have you tried changing the ultrasound frequency and examining the possibility of wifi interference?

sagerd, 2016-10-25:
@mani-monaj I'm pretty sure it's wifi interference, all of the ultrasound readings seem to be accurate.  From my understanding, it just seems like the wireless hardware used in the drone is just really bad and prone to dropping connections.  Is this your experience?  Do you have any suggestions?

sagerd, 2016-10-25:
Also, this is unrelated, but has anyone had better luck using two different computers and wifi dongles on different channels to get better connections?  What needs modified, I had two different computers on the same network and they interacted with each other while the drones were connected, but the drones never took off.

mani-monaj, 2016-12-07:
> Is this your experience?

Unfortunately yes.

A good wireless router (access point) makes a huge difference. Also you can change the channel manually to a non-crowded channel (i.e. using the Wifi Scanner on Android). What is your configuration like now?