AutonomyLab/ardrone_autonomy #196: Installing ardrone_autonomy in Kinetic

bermejocrew, 2016-09-08:
Hi everyone,
I'm new in ROS and i'm trying to install ardrone_autonomy package to Kinetic version. I tried to install it with the guide  but it doesnt work. I tried to execute

 

> apt-get install ros-hydro-ardrone-autonomy

it didnt work. Someone cal help me to install it??

Thank you

pulver22, 2016-09-08:
The latest version supported should be Jade (I think). I'm still on Indigo and no issues at all!

bermejocrew, 2016-09-09:
And how did you install it in ROS Indigo??

pulver22, 2016-09-09:
With the command you use above just substituting hydro with indigo

mani-monaj, 2016-10-14:
`ardrone_autonomy` is available on ROS *Kinetic*. Have you tried running `apt-get install ros-kinetic-ardrone-autonomy`?

ktiwari9, 2017-06-18:
Hello @mani-monaj ,
I am having trouble running ardrone_autonomy on ROS Kinetic. I did `sudo apt-get install ros-kinetic-ardrone-autonomy` like you said above. I then did `source /opt/ros/kinetic/setup.bash && roslaunch ardrone_autonomy ardrone.launch`. When I do rostopic list, I dont see the gps data being published. Could you please help me out here ?

farzi2007, 2019-04-24:
Did anybody get a solution to the question posted by @ktiwari9 ?