AutonomyLab/ardrone_autonomy #197: How to set frame rate of bottom camera to 60fps?

YuxiHuang001, 2016-09-30:
Hi, I am using AR.Drone 2.0 and using ardrone_autonomy, when I use "rosservice call /ardrone/togglecam" to change the camera from front camera to bottom camera, but when I subscribe the /ardrone/image_raw topic I found that the frame rate of bottom camera is only 20HZ, but I asked the ardrone technical support ,he said that the frame rate of bottom camera can be 60fps, could you tell me how to set? Thx a lot

mani-monaj, 2016-10-14:
@YuxiHuang001 I have never came across this feature. Has the technical support person provided any technical documentation or hints? Have they mentioned that it is exposed as an SDK feature?

raultron, 2017-01-13:
wow, 60 fps? I am also interested to know the answer to this.

mani-monaj, 2017-02-08:
@YuxiHuang001 Any updates on this?

I will close this issue for now. Please consider re-opening it if you have more information.

closed 2017-02-08 by mani-monaj