AutonomyLab/ardrone_autonomy #198: Extracting stamp info

cyberphantom, 2016-10-03:
Hi,
I'm using ardrone_autonomy and tum_ardrone to get /ardrone/image_raw and /ardrone/predictedPose and record the fields of these topics to a bag file. Then  I extracted the frames to a folder using 
rosrun image_view extract_images _sec_per_frame:=0.01 image:=/ardrone/image_raw
and then I extracted the Positions using
rostopic echo -b flight.bag -p /artdrone/predictedPose> predictedPose.txt
Now I want to match the positions with their closest frames. The problem is I can't get the stamp of the header of the image_raw topic. It says the header info in stamp field. can anyone please give a solution on matching the records of these two topics?

Thanks

mani-monaj, 2016-10-14:
I suggest that you post this question to [ROS Answers: Open Source Q&A Forum](http://answers.ros.org/questions/)

mani-monaj, 2016-12-07:
@cyberphantom Any update on this issue?

cyberphantom, 2016-12-07:
Yes, It seams that if you tried to read the header information by executing ROS command, you can't reach to timestamp info, but if you use any sort of python or cpp coding to access the header information you can get the timestamp header info.

Thanks for following up!

mani-monaj, 2016-12-07:
Thanks for the update. I will close this issue then.

closed 2016-12-07 by mani-monaj