AutonomyLab/ardrone_autonomy #200: No bottom camera stream

JBot, 2017-01-09:
I have recently bought an ArDrone2 and I am trying to use this package to control it and see the different cameras.

If I use the freeflight software everything is ok, I can see the bottom camera.

But, if I run the standard launchfile I can't see any video from the bottom camera (the front camera work well) and if I use again freeflight without removing the drone battery, freeflight doesn't show the bottom camera.

My drone version is 2.2 and the software is 2.4.8. 
Do you have and idea of the reason ?

raultron, 2017-01-13:
I think it is stated in the documentation somewhere that front camera is the default, you have to enable the bottom camera using the service provided for that. In the command line:

`rosservice call /ardrone/setcamchannel "channel: 1"`

mani-monaj, 2017-02-08:
Thanks @raultron for your answer.

@JBot I close this issue for now since the answer should have solved your problem. Please consider re-opening the issue if there are still any issues.

closed 2017-02-08 by mani-monaj