AutonomyLab/ardrone_autonomy #202: can't build SDK

sezan92, 2017-01-21:
Everytime I type ./build_sdk.sh the terminal shows no such file has been found. My OS is Ubuntu 14.04 and I am using Ros indigo

SupermanPrime01, 2017-01-29:
[New Software Guide 01-28-17.pdf](https://github.com/AutonomyLab/ardrone_autonomy/files/737402/New.Software.Guide.01-28-17.pdf)

I had some trouble following the tutorials provided by Mike Hamer because he was using Fuerte, but I tried my best to record my steps to get the AR drone working on indigo. Attached is a document I wrote on what I did. Perhaps it may help you.

mani-monaj, 2017-02-08:
You don't need to run `build_sdk.sh` anymore to compile the driver. All the required steps to compile to driver from source is provided here: http://ardrone-autonomy.readthedocs.io/en/latest/installation.html

However, you can install the driver on ROS Indigo using the binary installation method: `$ sudo apt-get install ros-indigo-ardrone-autonomy`.

`ardrone_tutorials` package is quite old and is not compatible with ROS Indigo as far as I can tell. You may be able to browse through its forks and see if you can find a more up-to-date version: 

https://github.com/mikehamer/ardrone_tutorials/network