AutonomyLab/ardrone_autonomy #204: Academy download paused

cardcaster, 2017-02-09:
Hi,
I'm using Ubuntu 16.04 LTS on Virtual box. Whenever I connect my Ar Drone 2.0 via its ad-hoc network and run the command " rosrun ardrone_autonomy ardrone_driver" on the terminal this is the output i get

`
===================+> 192.168.1.1
[ INFO] [1486597148.554224001]: No values found for `~cov/imu_la` in parameters, set all to zero.
[ INFO] [1486597148.555073608]: No values found for `~cov/imu_av` in parameters, set all to zero.
[ INFO] [1486597148.555873477]: No values found for `~cov/imu_or` in parameters, set all to zero.
[ WARN] [1486597148.559227177]: realtime navdata is off, odometry may be imprecise
Deleting Profile -cea48d22
Deleting Application -e182b69b
Input device Teleop added
Starting thread video_stage
Starting thread video_recorder
Starting thread update_ros
Set IP_TOS ok
Starting thread navdata_update
Starting thread ardrone_control
Thread `update_ros` started 
 Video recorder thread initialisation
PA : MEMORY SPACE ALLOWED : 40 MB

video stage thread initialisation

Video multisocket : init 2 sockets
Video multisocket : connecting socket 0 on port 0 UDP
Video multisocket : connecting socket 1 on port 5555 TCP
Thread navdata_update in progress...
Start thread thread_academy_download 
Academy download stage resumed
Start thread thread_academy_upload
Start thread thread_academy
Timeout when reading navdatas - resending a navdata request on port 5554
Sending default CAT_APPLI settings
Sending default CAT_USER settings
Sending default CAT_SESSION settings
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Academy download stage resumed
[ INFO] [1486597153.646525080]:   SEND: CAT_APPLI/navdata_options = 268435456.000000 (DEFAULT = 65537.000000)
[ INFO] [1486597153.646743869]:   SEND: CAT_SESSION/video_codec = 129.000000 (DEFAULT = 32.000000)
[ INFO] [1486597153.647876027]:   SEND: CAT_SESSION/max_bitrate = 4000.000000 (DEFAULT = 1000.000000)
[ INFO] [1486597153.720829555]: Successfully connected to 'My ARDrone' (AR-Drone 2.0 - Firmware: 2.4.8) - Battery(%): 79
[ INFO] [1486597153.721010163]: Navdata Publish Settings:
[ INFO] [1486597153.721384751]:     Legacy Navdata Mode: On
[ INFO] [1486597153.721709536]:     ROS Loop Rate: 50 Hz
[ INFO] [1486597153.721996501]:     Realtime Navdata Publish: Off
[ INFO] [1486597153.722180974]:     Realtime Video Publish: `Off`
[ INFO] [1486597153.722358548]:     Drone Navdata Send Speed: 200Hz (navdata_demo=0)
Academy download stage paused
exit
^X^CShutting down ... 
 ROS loop terminated ... 
Input device Teleop removed
Academy download stage resumed

video stage thread ended

Custom ardrone tool ended
`
No matter how many times I try to reconnect it, it always stops at "Academy download stage  paused". Please if anyone knows the cause of this problem, do let me know. Thank you.

mani-monaj, 2017-02-10:
Have you tried running the driver in a non-VM environment?

cardcaster, 2017-02-12:
Thank you for replying. No I haven't tried running Ubuntu in a non-VM on my MacBook Pro. I do not want to dual boot my laptop as I only need Ubuntu for this project. Sorry, are there any other solutions on running Ubuntu other than VM. Maybe bootable USB/CD?

pulver22, 2017-02-13:
> Maybe bootable USB/CD?

Yes, you can install ubuntu on a USB and run it everytime from there. It's a different solution from the live boot because you can also save data on the usb. If interested, please search for it online. Here it's not the proper place !

cardcaster, 2017-02-13:
Thank you so much. I'll go with the usb solution and I will let you know if I'm still facing the same issue.

mani-monaj, 2017-02-18:
Thanks @pulver22 for your answer.

@cardcaster I will close this issue. Please feel free to re-open it if you have further questions/updates.

closed 2017-02-18 by mani-monaj

cardcaster, 2017-02-20:
Hi,
Sorry for the late reply. This is the output I get on non-vm ubuntu:

h@h-MacBookPro:~$ rosrun ardrone_autonomy ardrone_driver 
Wait authentification
Wait authentification
Wait authentification
===================+> 192.168.1.1
[ INFO] [1487559963.209763307]: No values found for `~cov/imu_la` in parameters, set all to zero.
[ INFO] [1487559963.210315056]: No values found for `~cov/imu_av` in parameters, set all to zero.
[ INFO] [1487559963.210861838]: No values found for `~cov/imu_or` in parameters, set all to zero.
[ WARN] [1487559963.213581180]: realtime navdata is off, odometry may be imprecise
Deleting Profile -cea48d22
Deleting Application -e182b69b
Input device Teleop added
Starting thread video_stage
Starting thread video_recorder

video stage thread initialisation

Starting thread update_ros
Video multisocket : init 2 sockets
Video multisocket : connecting socket 0 on port 0 UDP
Video multisocket : connecting socket 1 on port 5555 TCP
Video recorder thread initialisation
Thread `update_ros` started 
 Set IP_TOS ok
Starting thread navdata_update
Starting thread ardrone_control
Thread navdata_update in progress...
PA : MEMORY SPACE ALLOWED : 40 MB
Start thread thread_academy_download 
Academy download stage resumed
Start thread thread_academy
Start thread thread_academy_upload
Academy download stage paused
Timeout when reading navdatas - resending a navdata request on port 5554
Academy download stage resumed
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
(repeats again)

Sending default CAT_APPLI settings
Sending default CAT_USER settings
Sending default CAT_SESSION settings
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
(repeats again)
[ INFO] [1487559968.238890594]:   SEND: CAT_APPLI/navdata_options = 268435456.000000 (DEFAULT = 65537.000000)
[ INFO] [1487559968.239092491]:   SEND: CAT_SESSION/video_codec = 129.000000 (DEFAULT = 32.000000)
[ INFO] [1487559968.239389042]:   SEND: CAT_SESSION/max_bitrate = 4000.000000 (DEFAULT = 1000.000000)
[ INFO] [1487559968.276339423]: Successfully connected to 'My ARDrone' (AR-Drone 2.0 - Firmware: 2.4.8) - Battery(%): 52
[ INFO] [1487559968.276412544]: Navdata Publish Settings:
[ INFO] [1487559968.276471757]:     Legacy Navdata Mode: On
[ INFO] [1487559968.276527382]:     ROS Loop Rate: 50 Hz
[ INFO] [1487559968.276593876]:     Realtime Navdata Publish: Off
[ INFO] [1487559968.276656273]:     Realtime Video Publish: Off
[ INFO] [1487559968.276717200]:     Drone Navdata Send Speed: 200Hz (navdata_demo=0)
Academy download stage paused
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
Tag 28 is an unknown navdata option tag
Tag 29 is an unknown navdata option tag
(repeats again)

Timeout when reading navdatas - resending a navdata request on port 5554
Timeout when reading navdatas - resending a navdata request on port 5554
Reconnecting ... Reconnecting ... OK
Reconnecting ... OK
OK

Brunorcarneiro, 2017-03-25:
Hello, I have the same problem trying to running the example ardrone_navigation. Can someone help us??

HALtheWise, 2017-09-23:
I initially had the same issue while running ardrone_autonomy in an Ubuntu VM, but ultimately got it to work after disconnecting my phone from the drone's network.

Working configuration (for me):
- Guest OS: Ubuntu 16.04 with ROS Kinetic
- Host OS: Ubuntu 16.04 with VirtualBox
- Network mode: _either_ "bridged" _or_ "NAT"
- Other devices disconnected (wifi disabled)

Note that in bridged mode, the guest OS is assigned an IP by the drone that is different from the one used by the host, while in NAT mode it is assigned an IP address by the host that is not in the 192.168.* block. Either way seems to work fine.