AutonomyLab/ardrone_autonomy #208: Kinetic installation without nodes

drc94, 2017-04-22:
Hi,

I have installed ardrone_autonomy using $ sudo apt-get install ros-kinetic-ardrone-autonomy and it is in the following folder: 
/opt/ros/kinetic/share/ardrone_autonomy

But when I try to execute the launch file using $ roslaunch ardrone_autonomy ardrone.launch it gives me the error: 
ERROR: cannot launch node of type [ardrone_autonomy/ardrone_driver]: can't locate node [ardrone_driver] in package [ardrone_autonomy]

Looking in the package folder I can see that there is only the cmake, launch, msg and srv folders, so there is not the src folder where should be the nodes (at least the ardrone_driver). I'm working into ROS Kinetic. I tried to remove and install again the package but it doesn't change. Could someone help me? 

Thanks.

pulver22, 2017-04-24:
Hi @drc94,
after installing the package did you try to resource /opt/ros/kinetic/setup.bash ?

drc94, 2017-04-27:
thanks @pulver22, yes I did

ktiwari9, 2017-06-16:
I am having the same problem for ros Kinetic. Any suggestions how to fix it ?

vinbo, 2017-06-27:
i met the same problem under version indigo, can anyone help?

SohamJani, 2017-09-20:
Any updates? Same issue here

shubhammashere, 2018-03-21:
i have resourced the packages, the file ardrone_driver.h is existing in the include folder but still its giving the same error