AutonomyLab/ardrone_autonomy #217: Invoking "make -j4 -l4" failed on ROS Kinetic

SrijaneeBiswas, 2018-01-26:
Hi All,

I'm trying to install the package on ROS Kinetic. When I try to catkin_make after git cloning the package, it fails with the error message as mentioned in the title. Is the package compatible only with ROS versions below Kinetic? If so, can you kindly provide a link to install ardrone_autonomy for ROS Kinetic ?

Thanks,
Srijanee Biswas

pulver22, 2018-01-26:
Hi @SrijaneeBiswas,

you can simply install the driver using sudo apt install `ros-kinetic-ardrone-autonomy`.

SrijaneeBiswas, 2018-01-26:
After doing that, I tried doing "rosrun ardrone_autonomy ardrone_driver" but first of all, it wouldn't tab-complete (meaning the package is not recognized) and after executing the command, it throws up an error "Couldn't find executable ardrone_driver". Any ideas why that might be the case?

pulver22, 2018-01-26:
Try to resource /opt/ros/kinetic/setup.bash, but you should not have that problem.

SrijaneeBiswas, 2018-01-28:
Thanks, it worked after sourcing.