AutonomyLab/ardrone_autonomy #221: Installing ardrone_autonaomy and tum simulator on Kinetic

caesar84, 2018-04-09:
Hi there I am trying ti install ardrone_autonomy and tum_simulator on ROS kinetic and i had faced different problems each time i try:
1- since i have catkin already installed then i followed the tutorial http://ardrone-autonomy.readthedocs.io/en/latest/installation.html#compile-from-source:
`$ cd ~/catkin_ws/src
$ git clone https://github.com/AutonomyLab/ardrone_autonomy.git -b indigo-devel
$ cd ~/catkin_ws
$ rosdep install --from-paths src -i
$ catkin_make`
then error came (because i already had Hector_quadrotor installed on catkin_ws:
`Base path: /home/mastlab/catkin_ws
Source space: /home/mastlab/catkin_ws/src
Build space: /home/mastlab/catkin_ws/build
Devel space: /home/mastlab/catkin_ws/devel
Install space: /home/mastlab/catkin_ws/install
Multiple packages found with the same name "message_to_tf":
- hector_localization/message_to_tf
- tum_simulator/message_to_tf`

2- i followed different tutorial (https://github.com/iolyp/tum-simulator-indigo) by creating a separated work space as follow:
`mkdir -p ~/tum_simulator_ws/src
cd  ~/tum_simulator_ws/src
catkin_init_workspace
git clone https://github.com/AutonomyLab/ardrone_autonomy.git	
git clone https://github.com/iolyp/tum-simulator-indigo.git
cd ..
rosdep install --from-paths src --ignore-src --rosdistro indigo -y` then when i ran
 `catkin_make
` i face this error:
`tum-simulator-indigo/cvg_sim_gazebo_plugins/CMakeFiles/diffdrive_plugin_6w.dir/build.make:62: recipe for target 'tum-simulator-indigo/cvg_sim_gazebo_plugins/CMakeFiles/diffdrive_plugin_6w.dir/src/diffdrive_plugin_6w.cpp.o' failed
make[2]: *** [tum-simulator-indigo/cvg_sim_gazebo_plugins/CMakeFiles/diffdrive_plugin_6w.dir/src/diffdrive_plugin_6w.cpp.o] Error 1
CMakeFiles/Makefile2:6184: recipe for target 'tum-simulator-indigo/cvg_sim_gazebo_plugins/CMakeFiles/diffdrive_plugin_6w.dir/all' failed
make[1]: *** [tum-simulator-indigo/cvg_sim_gazebo_plugins/CMakeFiles/diffdrive_plugin_6w.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j6 -l6" failed
`

pulver22, 2018-04-10:
For what I know, tum_simulator is not compatible with Kinetic as it is in the official repository. But you can find a porting [here](https://github.com/pulver22/QLAB).

sezan92, 2018-09-16:
[this repo](https://github.com/eborghi10/AR.Drone-ROS ) helped me solve the problem

wajeehulhassanvii, 2019-03-29:
> [this repo](https://github.com/eborghi10/AR.Drone-ROS) helped me solve the problem

@sezan92 this repo worked for me but how can we launch other environments?

issouker97, 2020-05-23:
Hi @caesar84  i had followed the same website t install ardrone_autonomy i didn't get any problem everything  performed successfully then when i tried to install tum_simulator from this website https://github.com/iolyp/tum-simulator-indigo
i got almost the same problem , i would be grateful if you could check this problem in this link where i published it https://github.com/tum-vision/tum_simulator/issues/18
and give me your respectful and honorable opinion thanks in advanced see you in the comment