AutonomyLab/ardrone_autonomy #225: Accuracy of Navdata and controlling the drone through velocity command.

sinamr66, 2018-08-01:
Hi All,

I wonder if anyone has an estimation of the accuracy of Navdata for indoor environments, in particular velocities and accelerations. Moreover, how accurate the drone can be controlled through the velocity command? Is there a big difference between indoor and outdoor situations?
I don't need to know the exact value, an order of magnitude would be enough 😃 
Thanks a lot in advance. 
Best,
Sina

kbogert, 2018-08-02:
The short answer is: not very accurate.  I've never found dead-reckoning to be useful on these drones as there is just too much noise, you need to use some sort of state-estimation.

For the velocity command control, see #116 .  I implemented a fairly accurate control scheme in the uga_tum_ardrone package (https://github.com/thinclab/uga_tum_ardrone) based on a specialized PID controller (https://github.com/thinclab/uga_tum_ardrone/blob/732b3a132a0b35e42f350d09d31a26643eea5422/src/autopilot/DroneController.cpp#L148).

I would suggest trying that package first to see if it meets your needs before trying to do control yourself.

sinamr66, 2018-08-02:
Thanks a lot for the reply. It was indeed very helpful. :+1:  :+1: 
Just to clarify, your control works in indoors environment and there is no need to external measurements, am I right?

kbogert, 2018-08-02:
I have controlled the drones in both indoor and outdoor environments with uga_tum_ardrone.  I'm not sure what you mean by external measurements, the package does use the front camera to detect features in the environment in order to estimate its pose.

sinamr66, 2018-08-02:
It is clear now.... Thanks a lot
I meant, ,measuring the position of the drone by using external sensors, e.g., pozyx or Optitrack.

raultron, 2018-08-03:
If you want to rely only on internal measurements (not ptam for example) I implemented a very simple PID velocity controller for the Ardrone some time ago: 
https://github.com/raultron/ardrone_velocity

Later, Tobias Tüylü during his master thesis in our institute tried to improve it by including a state estimator: 
https://github.com/raultron/ardrone_velocity_ekf

The last one works with internal measurements with the possibility of integrating external measurements into the state estimator.

The biggest challenge for us was the delays in wifi communication which are hard to deal with in the controller since the delays are not constant. We found that the internal odometry measurements are quite good but with the Wifi delay is hard to do real-time control.